adafruit/Adafruit_NeoMatrix #5: keywords.txt missing?

eskern, 2016-11-25:
I've been trying to tinker with one of your snazzy NeoMatrix Shields (5X8) for a couple of days now. After reinstalling Arduino (updated to ver 1.6.13) I've installed the libraries for NeoMatrix, NeoPixel, and GFX — all the newest iterations. Upon testing out the matrixtext code there seems to be a processing problem of sorts. While NeoPixel library turns orange, NeoMatrix and GFX do not. Thus the NeoMatrix object doesn't seem to respond and the code doesn't run on the matrix shield (the "HOWDY" message doesn't show up). 

The Pixel library folder included a keywords.txt file. Matrix and GFX appear to be missing a similar document. What should I do to fix this? I was really looking forward to playing around with the matrix but can't do much at the moment.

[image]

PaintYourDragon, 2016-11-25:
keywords.txt is an optional thing that adds syntax highlighting in the editor; it's not harmful if missing, just won't provide orange-highlighted function calls, as you've observed. Very few of our libraries provide this as it's a much lower priority than tech support, new products, etc.

The NeoMatrix examples should work. In fact I just tested on one here, same version of Arduino IDE and all, seems to be working fine.

Two things to check for:
1. Has the external power trace on the PCB been cut? (It should be near the Ext 5V jack on current NeoPixel shields, or between Vin & Analog 0 on early models.) If the trace was previously cut, you'll need to bridge this if you want it all powered from USB.
2. Is this an RGBW (vs RGB) shield? If so, change NEO_GRB in the matrix declaration to add W.

eskern, 2016-11-26:
Thanks for clarifying about the .txt. The power trace is uncut, and the shield itself says that it's RGB. Perhaps something's the matter with my circuit then. I used the 5V and GND pins on arduino, connected to a breadboard with the shield. Added a 330 resistor to be safe and plugged the two into the respective holes on the shield that denote 5V and GND. Then had pin 6 connected to the shield's Din hole. Is this correct? Apologies for bringing hardware into this thread when I'm sure you want to stick to programming.

ladyada, 2016-11-26:
tech support is handled in the forums only.
please post there! thanks :)

closed 2016-11-26 by ladyada