adafruit/Adafruit_NeoMatrix #10: Want to Scroll Array Plz Help

LedArts, 2018-01-04:
Hello  Dear Master,
i want to scroll an Array in this code plz help me

#include <Adafruit_NeoPixel.h>
#define PIN 6           
#define MAX_LED 64       
uint32_t  LEDColour;
unsigned char x = 0;
unsigned char Colour[3];
unsigned char heart[10] = {12, 13, 17, 20, 26, 29, 33, 36, 44, 45};

Adafruit_NeoPixel strip = Adafruit_NeoPixel( MAX_LED, PIN, NEO_RGB + NEO_KHZ800 );
void setup() {
  // put your setup code here, to run once:
  strip.begin();
  // 
  strip.show();
}

void loop() {

  for (char number = 0; number < 64; number++)  //
    strip.setPixelColor(number,  0);
  strip.show();//
 
 // delay(1000);
  for (char number = 0; number < 10; number++)  //
    strip.setPixelColor(heart[number],  255,0,0); // i want to scroll that heart array
  strip.show();//
  delay(1000);
}

PaintYourDragon, 2018-01-05:
Github 'issues' are for reporting bugs. Please use the Adafruit Forums (forums.adafruit.com) for programming assistance, I'd suggest the 'Glowy things' topic. Please don't spam across multiple topics there either, it's no more likely to get a response, usually the opposite.

closed 2018-01-05 by PaintYourDragon

LedArts, 2018-01-05:
ok but its very small issue 
only tell me any code example to scroll it