adafruit/Adafruit_NeoMatrix #21: Led Color / ws2812b

whoo, 2020-12-20:
Hi,

I've got some issue with color functions.
The result is not the same ... and I don't know why.
(I try to play with brightness | color minimize light with it).

But cannot find the same color. (probably the 24bit and 32bit)
(What is the right way to deal with color with my ws2812b matrix)

```C
 Adafruit_NeoMatrix matrix = Adafruit_NeoMatrix(32, 8, 3,
  NEO_MATRIX_TOP    + NEO_MATRIX_LEFT +  NEO_MATRIX_COLUMNS + NEO_MATRIX_ZIGZAG ,     /// ZIGZAG for matrix display 
  NEO_RGB            + NEO_KHZ800);
  matrix.setPixelColor(1,0xff7e00);
  matrix.drawPixel(6,7,matrix.Color(0xff,0x7e,00));
  matrix.show();
```

Thanks.

ladyada, 2020-12-20:
hi please be more specific of what you're expecting and what you're seeing

whoo, 2020-12-20:
Hi,
Is it supposed to be the same color ?

```C
for (int x=0;x<8;x++)  matrix.setPixelColor(x+(5)*8,0x5f0e00);
for (int x=0;x<8;x++)  matrix.drawPixel(6,x,matrix.Color(0x5f,0x0e,00));
```

[image]

ladyada, 2020-12-20:
@PaintYourDragon might know

PaintYourDragon, 2020-12-21:
This is normal. setPixelColor() is a **NeoPixel library** function and works with linear 24-bit color. drawPixel() is a **GFX library** function and works with 16-bit ("565") gamma-corrected colors. The two should not be mixed unless you’re prepared for the outcome. It’s explained a little at the end of this page:

https://learn.adafruit.com/adafruit-neopixel-uberguide/neomatrix-library

There’s a hacky workaround matrix.setPassThruColor() function that lets you set uncorrected 24-bit colors for subsequent GFX calls, but I don’t really recommend it.

closed 2020-12-21 by PaintYourDragon