adafruit/Adafruit_NeoMatrix #22: neomatrix in ESP32-S2 am having difficulty

JohnK48, 2022-03-22:
I am building a NEOPIXEL panel (120x15 eventually, but now it is 120x8).  I also have two 16x16 panels to test with. The small panels seem to work properly using an Arduino Uno, but it does not have enough memory for the big panel. I have two programs working using the ESP32-S2, but noticed that the program that uses a single panel (and it works!) is coded wrong.  It is coded for the starting point to be NEO_MATRIX_TOP + NEO_MATRIX_LEFT, but the panel is wired for top right.

- Arduino board:  ESP32-S2 (Saola-1)
- Arduino IDE version (found in Arduino -> About Arduino menu):  2.0.0-rc3
 
- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**
// This works calling for ATMegaZero ESP32-S2
// This works, but should not.
// 
#include <Adafruit_GFX.h>
#include <Adafruit_NeoMatrix.h>
#include <Adafruit_NeoPixel.h>
//#ifndef PSTR
// #define PSTR // Make Arduino Due happy
//#endif
#define LED_PIN 18
#define MATRX_WIDTH 16
#define MATRX_HEIGHT 8
Adafruit_NeoMatrix matrix = Adafruit_NeoMatrix(
    MATRX_WIDTH, MATRX_HEIGHT, LED_PIN,
    NEO_MATRIX_TOP + NEO_MATRIX_LEFT + NEO_MATRIX_ROWS +
    NEO_MATRIX_ZIGZAG + 
    NEO_GRB + NEO_KHZ800);
const uint16_t colors[] = {
  matrix.Color(255,255,255),matrix.Color(255, 0, 0), matrix.Color(0, 255, 0), matrix.Color(0, 0, 255) };

void setup() {
  matrix.begin();
  matrix.setTextWrap(false);
  matrix.setBrightness(5);
  matrix.setTextColor(colors[0]);
}
int x    = matrix.width();
int pass = 0;

void loop() {
  matrix.fillScreen(0);
  matrix.setCursor(x, 0);
  matrix.print(F("Hello World"));
  if(--x < -36) {
    x = matrix.width();
    if(++pass >= 4) pass = 0;
    matrix.setTextColor(colors[pass]);
  }
  matrix.show();
  delay(100);
}

ladyada, 2022-03-22:
which adafruit neopixel products are you using?

JohnK48, 2022-03-23:
By this I presume you mean which LED strips? They are ws2812b strips 60 leds per meter, 5 meters long from BTF-LIGHTING. I have 6 strips, cut to 2 meter lengths, leaving 6 one meter strips put together for 3 more 2 meter rows. All in zigzag connection starting at upper left. I have two sets of power wires at about 30 inches from left and 50 inches from left. Intended to have two more power busses, but wanted to test first. I have a 5 volt power supply but have a battery pack for testing and for now have been powering from usb.There is another neomatrix library that claims better esp32 support. Is that worth trying?Regards,JohnSent from my T-Mobile 4G LTE Device

  difficulty (Issue #22)

ladyada, 2022-03-23:
yeah we dont provide this kind of tech support on github, please post to a forum!

closed 2022-03-23 by ladyada