adafruit/Adafruit_NeoMatrix #27: Esp32 core crashes in Adafruit NeoMatrix matrixtest example sketch (32x8 matrix panel)

vagvol, 2024-07-15:
Board
ESP32 Dev Module

Device Description
ESP32 DEVKIT V1 DOIT

Hardware Configuration
No

Core Version
v3.0.2

IDE Name
Arduino IDE 2.3.2

Operating System
Windows 11

Flash frequency
80Mhz

PSRAM enabled
no

Upload speed
921600

**Issue Description**
When I set Pixel Width to 32 and Pixel Height to 8, I have continuous reset loop. That issue does not occur with esp32 Core vesrsion 2.0.17. I provide you the crash error output:

13:31:48.570 -> rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:31:48.570 -> configsip: 0, SPIWP:0xee
13:31:48.570 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:31:48.570 -> mode:DIO, clock div:1
13:31:48.570 -> load:0x3fff0030,len:1448
13:31:48.570 -> load:0x40078000,len:14844
13:31:48.570 -> ho 0 tail 12 room 4
13:31:48.570 -> load:0x40080400,len:4
13:31:48.570 -> load:0x40080404,len:3356
13:31:48.570 -> entry 0x4008059c
13:31:49.367 -> ets Jul 29 2019 12:21:46

You don't need to attach any hardware to esp32. You can reproduce that issue by uploading "matrixtest" sketch from library examples. You should first change pixel width to 32 and height to 8.

vagvol, 2024-07-15:
I think that the problem is related to NeoPixel library (The NeoMatrix inherits NeoPixel class). I made some tests using the "simple" sketch included in NeoPixel library examples and found that setting a number for neopixels greater than 75 (eg.100pixels) causes crash loop.
```
// Which pin on the Arduino is connected to the NeoPixels?
#define PIN        5 // On Trinket or Gemma, suggest changing this to 1

// How many NeoPixels are attached to the Arduino?
#define NUMPIXELS 100 // Popular NeoPixel ring size
```
I hope that helps identifying the problem.

thomergil, 2024-07-21:
Confirming this problem. I have the same problem with a 16x16 LED matrix. Same error as @vagvol described.

maxelle, 2024-08-03:
Confirming the problem!

**With 8x8 everithing ok**
Adafruit_NeoMatrix matrix = Adafruit_NeoMatrix(8, 8, 1, 1, PIN,
  NEO_TILE_TOP   + NEO_TILE_LEFT   + NEO_TILE_ROWS   + NEO_TILE_PROGRESSIVE +
  NEO_MATRIX_TOP + NEO_MATRIX_LEFT + NEO_MATRIX_ROWS + NEO_MATRIX_ZIGZAG,
  NEO_GRB + NEO_KHZ800);

**With 16x16 crash!**
Adafruit_NeoMatrix matrix = Adafruit_NeoMatrix(**16, 16,** 1, 1, PIN,
  NEO_TILE_TOP   + NEO_TILE_LEFT   + NEO_TILE_ROWS   + NEO_TILE_PROGRESSIVE +
  NEO_MATRIX_TOP + NEO_MATRIX_LEFT + NEO_MATRIX_ROWS + NEO_MATRIX_ZIGZAG,
  NEO_GRB + NEO_KHZ800);

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1448
load:0x40078000,len:14844
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3356
entry 0x4008059c

maxelle, 2024-08-04:
Problem seems come from neopixel lib that allocates the led cache in stack (look esp.c function espShow).
Solved using a dedicated task with a very big stack size. 
Not the best solution but if you have enough memory it works and projects can continue...

egnor, 2024-08-05:
I think this is the relevant Adafruit_NeoPixel bug: https://github.com/adafruit/Adafruit_NeoPixel/issues/375

thomergil, 2024-09-26:
This problem persists in 3.0.5.