adafruit/Adafruit_NeoMatrix #28: Request for License Change from GPL-3.0 to More Permissive License

audi0615, 2024-07-23:
Hi.

Please change the current GPL-3.0 license of your library to a more permissive license like MIT or BSD. This change would allow us to integrate your library into our proprietary app without needing to open-source our code.

Thanks.

dhalbert, 2024-07-23:
Note that the license is LGPL-3.0, not GPL-3.0. LGPL does not require disclosing the rest of your source code. It has other requirements which are "Lesser". The requirements for an embedded product are subject to interpretation and I'm not going to interpret them at the moment.

The original commit that chose that license is here: https://github.com/adafruit/Adafruit_NeoMatrix/commit/d6f2bb3f57f0479c558b66aeea140161c2acc278. Note that https://github.com/adafruit/Adafruit_NeoPixel is also LGPL, perhaps because it was based on previous LGPL code.

@ladyada Do you remember why these are LGPL?

audi0615, 2024-07-23:
Thanks for your clarification. However, the license specified in GitHub is still GPL-3.0. Could you please revise it to be LGPL-3.0?

[image]

dhalbert, 2024-07-23:
Thanks for pointing that out. The license selector for GitHub does not include LGPL-3.0 as an easy choice. However, it appears that adding `COPYING.LESSER` will fix that.
https://github.com/licensee/licensee/issues/143
https://github.com/github/choosealicense.com/issues/607
The GitHub assistance on this is kind of a mess right now!

dhalbert, 2024-07-23:
I merged #29, and now it says this:

[image]

Note that LGPL-3.0 is an addendum to GPL-3.0. It is not a standalone license. I can also add SPDX lines to the source code, but the source code already says LGPL, not GPL. Is this OK for you now? I'll continue to inquire about why LGPL was chosen as the original license.

audi0615, 2024-07-23:
Thank you for your prompt action and for informing me that LGPL-3.0 is an addendum. 
Now everything is very clear.

ladyada, 2024-07-23:
ok do i have to do anything or is this resolved? :D

dhalbert, 2024-07-23:
All set, thanks!

closed 2024-07-23 by dhalbert